matlabbe/rtabmap_drone_example #2: Unable to take off

iamrajee, 2021-12-04:
I'm unable to take off the drone in the given gazebo world. Although, when I tried with the default gazebo simulation, it does take off.

# Drone does take off this way
Terminal1: roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
Terminal2: roslaunch px4 posix_sitl.launch
Terminal3: rosrun rtabmap_drone_example offboard

#But, doesn't take off this way:
Terminal 1:  roslaunch rtabmap_drone_example gazebo.launch
Terminal 2:  rosrun rtabmap_drone_example offboard

What could be the potential reason for this behaviour? Could it be because of the version miss-match of PX4/ mavros? I'm running this simulation on noetic.

# Here is output/error of gazebo.launch(terminal1):
```
... logging to /home/rajendra/.ros/log/366b5ef6-550c-11ec-b2d5-15a442893f81/roslaunch-u20-384705.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://u20:36799/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 566 lines cut ...]
    secs: 44
    nsecs: 170000000
  frame_id: ''
connected: True
armed: False
guided: False
manual_input: True
mode: "MANUAL"
system_status: 3

# rostopic echo /mavros/state (After running offboard)  
header: 
  seq: 217
  stamp: 
    secs: 217
    nsecs: 171000000
  frame_id: ''
connected: True
armed: False
guided: True
manual_input: False
mode: "AUTO.LAND"
system_status: 3

</details>

iamrajee, 2021-12-05:
When I tried with slam, then also it does not works:
Terminal 1: roslaunch rtabmap_drone_example gazebo.launch
Terminal 2: roslaunch rtabmap_drone_example slam.launch
Terminal 3: roslaunch rtabmap_drone_example rviz.launch
Terminal 4: rosrun rtabmap_drone_example offboard

See [this](https://youtu.be/VxXH8rsZV8s) video, where I describe the issue.

<details>
<summary>Log & Output(Expand to see.)</summary>

# Output Terminal1
```
rajendra@u20:~/drone_ws$ roslaunch rtabmap_drone_example gazebo.launch
... logging to /home/rajendra/.ros/log/366b5ef6-550c-11ec-b2d5-15a442893f81/roslaunch-u20-477300.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://u20:36613/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
[... 1089 lines cut ...]
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "2483"
      - 
        key: "Frequency (Hz)"
        value: "10.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.158679"
      - 
        key: "Last remote time (s)"
        value: "248.396000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.004203860"

</details>

matlabbe, 2021-12-09:
Here is my log on px4 terminal, it seems the flight software version is different. The EKF on your side is also not initialized.
```
[ INFO] [1639074645.906335780, 2.179000000]: GF: Using MISSION_ITEM_INT
[ INFO] [1639074645.906392478, 2.179000000]: RP: Using MISSION_ITEM_INT
[ INFO] [1639074645.906424167, 2.179000000]: WP: Using MISSION_ITEM_INT
[ INFO] [1639074645.906462046, 2.179000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1639074645.906490792, 2.179000000]: VER: 1.1: Flight software:     010c03ff (2e8918da66000000)
[ INFO] [1639074645.906522457, 2.179000000]: VER: 1.1: Middleware software: 010c03ff (2e8918da66000000)
[ INFO] [1639074645.906548027, 2.179000000]: VER: 1.1: OS software:         050b00ff (bf660cba2af81f05)
[ INFO] [1639074645.906574252, 2.179000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1639074645.906596910, 2.179000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1639074645.906620395, 2.179000000]: VER: 1.1: UID:                 4954414c44494e4f
INFO  [ecl/EKF] 4080000: EKF aligned, (baro hgt, IMU buf: 21, OBS buf: 15)
[ INFO] [1639074648.021382750, 4.291000000]: IMU: Attitude quaternion IMU detected!
INFO  [ecl/EKF] 4148000: EKF aligned, (baro hgt, IMU buf: 21, OBS buf: 15)
INFO  [ecl/EKF] 7080000: EKF aligned, (baro hgt, IMU buf: 21, OBS buf: 15)
INFO  [ecl/EKF] 7084000: EKF aligned, (baro hgt, IMU buf: 21, OBS buf: 15)
INFO  [ecl/EKF] 7084000: EKF aligned, (baro hgt, IMU buf: 21, OBS buf: 15)
INFO  [ecl/EKF] 7088000: EKF aligned, (baro hgt, IMU buf: 21, OBS buf: 15)
[ INFO] [1639074654.927322356, 11.171000000]: HP: requesting home position
INFO  [gyro_calibration] gyro 0 (1310988) updating calibration, [-0.0009, 0.0226, 0.0011] -> [0.0000, 0.0002, -0.0001] 11.8°C
INFO  [gyro_calibration] gyro 1 (1310996) updating calibration, [-0.0009, 0.0231, 0.0012] -> [0.0000, 0.0002, -0.0001] 11.8°C
INFO  [gyro_calibration] gyro 2 (1311004) updating calibration, [-0.0009, 0.0231, 0.0012] -> [0.0000, 0.0002, -0.0001] 11.8°C
INFO  [ekf2] 1 - resetting IMU bias
INFO  [ekf2] 2 - resetting IMU bias
INFO  [ekf2] 0 - resetting IMU bias
INFO  [ekf2] 4 - resetting IMU bias
INFO  [ekf2] 5 - resetting IMU bias
INFO  [ekf2] 3 - resetting IMU bias
[ INFO] [1639074659.968311303, 16.170000000]: GF: mission received
[ INFO] [1639074659.969410314, 16.172000000]: RP: mission received
[ INFO] [1639074659.969475302, 16.172000000]: WP: mission received
[ INFO] [1639074664.988477632, 21.172000000]: HP: requesting home position
```
What is your commit version of PX4-Autopilot version? Here is mine:
```
git log
commit 2e8918da66af37922ededee1cc2d2efffec4cfb2 (HEAD, tag: v1.12.3, origin/stable, origin/beta)
Author: Daniel Agar <daniel@agar.ca>
Date:   Mon Sep 6 11:52:58 2021 -0400

    boards: cubepilot_cubeorange only start ADSB mavlink if console not present
    
     - cubepilot_cubeorange_test has the console enabled (used for test rack and bench debugging)
```

iamrajee, 2021-12-12:
@matlabbe Thank you for your response, yes you're are right problem seems to be with a version mismatch. So I created a new workspace as per your instruction(thanks again for adding detailed instructions in Readme), and I could run it successfully.

closed 2021-12-12 by iamrajee

matlabbe, 2021-12-12:
Note that I just [updated](https://github.com/matlabbe/rtabmap_drone_example/commit/3134fbc600a4e5c48b33153e97eaa477c665eddc) the repo to use PX4 v1.12.3 on melodic too because of the losing altitude issue depicted in the overview video. If you are using PX4 v1.8.2, you will be required to update to PX4 v1.12.3 if you pull again this repo.