matlabbe/rtabmap_drone_example #3: Wondering where is the room model from.

twprinston, 2021-12-07:
Hi,

This is a great project for testing drone-slam in the simulated room world.

This is not the program issue, but I'm wondering where is the room model from.
Is the room model from scanned real world?
Could you teach how to make or get the room model? Any article hints are fine.

I'll really appreciate that. Thank you.

matlabbe, 2021-12-09:
The room model has been scanned with RTAB-Map Tango App https://play.google.com/store/apps/details?id=com.introlab.rtabmap
Sketchfab : https://skfb.ly/6Cpx8

If you have an iPhone or iPad with a LiDAR, you can use RTAB-Map for iOS App: https://apps.apple.com/ca/app/rtab-map-3d-lidar-scanner/id1564774365

The OBJ can be converted to collada using MeshLab, then a gazebo world/model can be created from that.

twprinston, 2021-12-13:
Got it.
Those are very helpful references.
Thank you so much!

closed 2021-12-13 by twprinston