matlabbe/rtabmap_drone_example #12: Generating the custom world

aaravrav142, 2023-07-05:
Hi 
How do I generate my own simulation world? I have an iPad Pro Lidar. Is it okay to use the PLY file from the rtabmap iPad app to generate the map and then export it in the gazebo for use with a ground robot simulation? What steps do I need to follow? Can you please explain? Thanks

matlabbe, 2023-07-10:
You will have to convert the OBJ or PLY into a DAE file (collada). You can use MeshLab to do the conversion. You then need to create an SDF file including the DAE file for the scan in a [models](https://github.com/matlabbe/rtabmap_drone_example/tree/master/models) folder, then create a gazebo [world](https://github.com/matlabbe/rtabmap_drone_example/tree/master/worlds) file to include that model. To use paths in SDF file relative to package, edit package.xml and ad this:
https://github.com/matlabbe/rtabmap_drone_example/blob/eaeeb6f723cd0b0307575d3969f2cb5df42f928e/package.xml#L76-L82

For ground robot, to make sure the ground is flat, a trick is to add an invisible ground plane at z=0 and lower the model for about 1-2 cm.

matlabbe, 2024-06-16:
Just select DAE when exporting like in this video: https://youtu.be/37E1iHAJp24?si=1ODDCLea7IoAWQp1

tulingcheng86, 2024-06-17:
> Just select DAE when exporting like in this video: https://youtu.be/37E1iHAJp24?si=1ODDCLea7IoAWQp1

THANKS