matlabbe/rtabmap_drone_example #13: using vins fusion slam

youngjae41, 2023-07-10:
Hello, I am trying to use vio with vins fusion and mapping with rtabmap using this good code.

I downloaded the rtabmap source and set vinsFusion=On to complete the installation. But can you tell me the launch file to use this vins fusion as odom?

matlabbe, 2023-07-11:
You can look at this example:
https://github.com/introlab/rtabmap_ros/blob/c1d1c01d5d732784ceb6b225ebf6eebdb700ceb0/rtabmap_examples/launch/euroc_datasets.launch#L26-L32

```
Odom/Strategy 9 OdomVINS/ConfigPath ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml
```

To make sure if rtabmap did build with VINS, you can do `rtabmap --version`. If you see `With VINS-Fusion:        true`, the parameters in the example above should work.

youngjae41, 2023-07-12:
@matlabbe T

here is no problem with Vins and Rtabmap working normally.

The problem is that tf related errors occur.

matlabbe, 2023-07-14:
Which tf errors?

youngjae41, 2023-07-15:
@matlabbe 
I am testing the above algorithm on a real drone. I am currently using orangepi5, and at first it outputs 30hz odom and then gradually goes down. Are there options available for less powerful computers?

I also tried Vins fusion, but it takes a lot of resources

matlabbe, 2023-07-18:
For VIO, you may try [msckf_vio](https://github.com/KumarRobotics/msckf_vio) (see Table 7 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf)), which is made to use less computation power (i.e., higher framerate on less powerful computer) and accuracy is still relatively good. If you want to use it inside rtabmap, you may have to follow this:
https://github.com/introlab/rtabmap/blob/64f79813cd2c3a7e3f916a7ef2637e93f1f13a5f/docker/jfr2018/latest/Dockerfile#L140-L141
to build msckf in a catkin workspace, then do this:
```
cd rtabmap/build
cmake -DWITH_MSCKF_VIO=ON ..
```

youngjae41, 2023-07-19:
@matlabbe Thanks matlabbe 

Right now I'm trying to use the existing rtabmap slam and I'm testing it, but I've noticed something strange.

[image]
[image]
[image]

camera : d435i 
imu : /mavros/imu/data

I am using it as described above, and the tf of the base frame are all showing different directions.

my launch file

 <arg name="localization" default="false"/>
    <arg name="rtabmap_viz"   default="false"/>
    <arg name="ground_truth" default="false"/>

   <node pkg="tf" type="static_transform_publisher" name="base_link_imu" args="0 0 0 0 0 0 base_link imu_link 100" />

   <node pkg="tf" type="static_transform_publisher" name="base_link_camera" args="0.05 0 0 0 0 0 base_link camera_link 100"/>

    <arg     if="$(arg localization)" name="pre_args" default=""/>
    <arg unless="$(arg localization)" name="pre_args" default="-d"/>

    <node pkg="nodelet" type="nodelet" name="imu_to_tf" args="standalone rtabmap_util/imu_to_tf">
      <remap from="imu/data" to="/mavros/imu/data"/>
      <param name="fixed_frame_id" value="base_link_stabilized"/>
      <param name="base_frame_id" value="base_link"/>
    </node>

    
    <param name="/rtabmap/rtabmap/use_action_for_goal" value="true"/>
    <remap from="/rtabmap/move_base" to="/move_base"/>

    
    <param name="/rtabmap/rtabmap/latch" value="false"/> 
      <node pkg="nodelet" type="nodelet" name="camera_points_xyz" args="standalone rtabmap_util/point_cloud_xyz">
      <remap from="depth/image"       to="/camera/depth/image_rect_raw"/>
      <remap from="depth/camera_info" to="/camera/depth/camera_info"/>
      <remap from="cloud"             to="camera_cloud" />

      <param name="decimation"  type="double" value="4"/>
      <param name="voxel_size"  type="double" value="0.1"/>
      <param name="approx_sync" type="bool"   value="true"/>
    </node>

    
    <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
       <remap from="map" to="/rtabmap/grid_map"/>
       <remap from="odom" to="/rtabmap/odom"/>
       <param name="base_local_planner" value="dwa_local_planner/DWAPlannerROS" />
       <rosparam file="$(find rtabmap_drone_example)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
       <rosparam file="$(find rtabmap_drone_example)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
       <rosparam file="$(find rtabmap_drone_example)/param/global_costmap_params.yaml" command="load" />
       <rosparam file="$(find rtabmap_drone_example)/param/local_costmap_params.yaml" command="load" />
       <rosparam file="$(find rtabmap_drone_example)/param/base_local_planner_params.yaml" command="load" />
    </node>

Can you tell me what's wrong? thank you

youngjae41, 2023-07-19:
@matlabbe 

my tf tree
[image]

matlabbe, 2023-07-26:
What is the `frame_id` in `/mavros/imu/data`? If different than `base_link`, the 18- deg may come from `base_link` to `imu_link`, see `imu_to_tf` code here: https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_util/src/nodelets/imu_to_tf.cpp

youngjae41, 2023-07-27:
@matlabbe thanks matlabbe I am using the px4.config file provided by rtabmap_drone_example and imu is written as base_link

And as recommended by matlabbe, I am using it after changing it to MSCKF_VIO. When base_link TF rotates on the map, base_link_stabilized TF is fixed instead of rotating along with it.

matlabbe, 2023-07-30:
1. `Grid/MaxGroundHeight` is based on map frame (`Grid/MapFrameProjection`), assuming the drone is taking off from the floor. Did you compare the odometry values between both visual odometry approaches?
2. Well, for how much it will affect VIO you would need to ask on MSCKF repo.

youngjae41, 2023-07-31:
@matlabbe 

When testing all three of orb-slam , vins and msckf_vio , msckf_vio performed the best as recommended.

As with vins, I thought it would be a good way to achieve a more accurate vio to set each cam, imu, gyro, and acc using a config file.

Due to lack of CPU power, Z is using altitude lidar. Is there any way to integrate a separate lidar sensor?

Thank you for your help. When the code is complete, we will publish the code with a thank you note

matlabbe, 2023-08-05:
> Is there any way to integrate a separate lidar sensor?

Not sure where you want to integrate it. rtabmap node can subscribe to a lidar or Pointcloud2 topic.

youngjae41, 2023-08-08:
@matlabbe thanks

closed 2023-08-08 by youngjae41