matlabbe/rtabmap_drone_example #15: Hello, have you tried to replace the move_base with other 3D path planning methods？

tulingcheng86, 2024-06-17:
for example，use [egoplanner](https://github.com/ZJU-FAST-Lab/ego-planner)
 i want to chang move_base to egoplanner .

matlabbe, 2024-06-17:
We used a custom 3D planning approach for that [paper](https://www.researchgate.net/publication/362336640_TAPE_Tether-Aware_Path_Planning_for_Autonomous_Exploration_of_Unknown_3D_Cavities_using_a_Tangle-compatible_Tethered_Aerial_Robot), but the code was not released as open source. However the planner was not meant for real-time object avoidance like egoplanner, but more for exploration. In all cases, rtabmap provides the 6DoF pose of the drone, so you can replace move_base with a any planner. If egoplanner takes care of the 3D occupancy grid already, then you just need the output pose from rtabmap. rtabmap can produce an OctoMap online though if needed (at 1 Hz), but for fast flying, you would need a fast real-time updated local occupancy grid.

tulingcheng86, 2024-06-18:
> 我们在那[篇论文](https://www.researchgate.net/publication/362336640_TAPE_Tether-Aware_Path_Planning_for_Autonomous_Exploration_of_Unknown_3D_Cavities_using_a_Tangle-compatible_Tethered_Aerial_Robot)中使用了定制的 3D 规划方法，但代码并未作为开源发布。然而，规划器并不像自我规划器那样用于实时对象回避，而更多地用于探索。在所有情况下，rtabmap 都提供无人机的 6DoF 姿势，因此您可以用任何规划器替换move_base。如果 egoplanner 已经处理了 3D 占用网格，那么您只需要 rtabmap 的输出姿势。如果需要，rtabmap 可以在线生成 OctoMap（1 Hz），但对于快速飞行，您需要一个快速实时更新的本地占用网格。

Thanks for the idea, I'm going to try it