matlabbe/rtabmap_drone_example #17: Can this simulation environment be ported to ROS2?

tulingcheng86, 2024-07-22:


matlabbe, 2024-07-22:
Maybe. There are many things to consider:
 * The ros1 version is using specific PX4  and mavros versions, so the current px4 config or mavros connection may not work with latest code
 * The Gazebo model may need to be updated for latest gazebo (e..g, px4 simulated controller the same?)
 * nav2 config may be slithery different than nav stack on ros1, though should work like on ros1
 * rtabmap launch file be ported to ros2 launch file, otherwise it should work like on ros1

It feels that a new repo (or new branch called `ros2`?) may need to be created as most things would change.

matlabbe, 2025-10-29:
See [ros2](https://github.com/matlabbe/rtabmap_drone_example/tree/ros2?tab=readme-ov-file#install) branch