matlabbe/rtabmap_drone_example #19: Docker Build fails

guptaPrabhav, 2024-07-23:
Hello,

On folowing the instructions in the README, the docker build command fails halfway through. Here is the terminal output of what is happening. The error message is at the end of the code block, but there are a few warnings in between as well.

```
[+] Building 387.2s (10/14)                                                                                                                                                                           docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                            0.0s
 => => transferring dockerfile: 3.20kB                                                                                                                                                                          0.0s
 => [internal] load metadata for docker.io/library/ros:noetic-perception                                                                                                                                        0.3s
 => [internal] load .dockerignore                                                                                                                                                                               0.0s
 => => transferring context: 2B                                                                                                                                                                                 0.0s
 => [ 1/10] FROM docker.io/library/ros:noetic-perception@sha256:e4061cbfcbcf1367c0c7d1686dcbf19e74fd120065c6c05dc5f7b00a8169651f                                                                                0.0s
 => [internal] load build context                                                                                                                                                                               0.0s
 => => transferring context: 3.59kB                                                                                                                                                                             0.0s
 => CACHED [ 2/10] RUN apt update && apt install     ros-noetic-gazebo-dev     ros-noetic-joy     ros-noetic-teleop-twist-joy     ros-noetic-geographic-msgs     ros-noetic-dwa-local-planner     ros-noetic-r  0.0s
 => CACHED [ 3/10] RUN pip3 install numpy toml packaging jinja2 empy pyros-genmsg                                                                                                                               0.0s
 => CACHED [ 4/10] RUN rm /bin/sh && ln -s /bin/bash /bin/sh                                                                                                                                                    0.0s
 => CACHED [ 5/10] WORKDIR /root/                                                                                                                                                                               0.0s
 => ERROR [ 6/10] RUN git clone https://github.com/PX4/PX4-Autopilot.git &&    cd PX4-Autopilot &&     git checkout v1.12.3 &&     git submodule update --init --recursive &&     make -j8 px4_sitl_default   386.8s
------                                                                                                                                                                                                               
 > [ 6/10] RUN git clone https://github.com/PX4/PX4-Autopilot.git &&    cd PX4-Autopilot &&     git checkout v1.12.3 &&     git submodule update --init --recursive &&     make -j8 px4_sitl_default gazebo &&     make px4_sitl_default install &&     cd &&     rm -rf PX4-Autopilot:                                                                                                                                                   
0.197 Cloning into 'PX4-Autopilot'...                                                                                                                                                                                
49.12 Note: switching to 'v1.12.3'.                                                                                                                                                                                  
49.12                                                                                                                                                                                                                
49.12 You are in 'detached HEAD' state. You can look around, make experimental
49.12 changes and commit them, and you can discard any commits you make in this
49.12 state without impacting any branches by switching back to a branch.
49.12 
49.12 If you want to create a new branch to retain commits you create, you may
49.12 do so (now or later) by using -c with the switch command. Example:
49.12 
49.12   git switch -c <new-branch-name>
49.12 
49.12 Or undo this operation with:
49.12 
49.12   git switch -
49.12 
49.12 Turn off this advice by setting config variable advice.detachedHead to false
49.12 
49.12 HEAD is now at 2e8918da66 boards: cubepilot_cubeorange only start ADSB mavlink if console not present
49.21 Submodule 'Tools/flightgear_bridge' (https://github.com/PX4/PX4-FlightGear-Bridge.git) registered for path 'Tools/flightgear_bridge'
49.21 Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/jMAVSim'
49.21 Submodule 'Tools/jsbsim_bridge' (https://github.com/PX4/px4-jsbsim-bridge.git) registered for path 'Tools/jsbsim_bridge'
49.21 Submodule 'Tools/sitl_gazebo' (https://github.com/PX4/PX4-SITL_gazebo.git) registered for path 'Tools/sitl_gazebo'
49.21 Submodule 'boards/atlflight/cmake_hexagon' (https://github.com/PX4/cmake_hexagon.git) registered for path 'boards/atlflight/cmake_hexagon'
49.21 Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path 'mavlink/include/mavlink/v2.0'
49.21 Submodule 'platforms/nuttx/NuttX/apps' (https://github.com/PX4/NuttX-apps.git) registered for path 'platforms/nuttx/NuttX/apps'
49.21 Submodule 'platforms/nuttx/NuttX/nuttx' (https://github.com/PX4/NuttX.git) registered for path 'platforms/nuttx/NuttX/nuttx'
49.21 Submodule 'platforms/qurt/dspal' (https://github.com/ATLFlight/dspal.git) registered for path 'platforms/qurt/dspal'
49.21 Submodule 'src/drivers/gps/devices' (https://github.com/PX4/PX4-GPSDrivers.git) registered for path 'src/drivers/gps/devices'
49.21 Submodule 'src/drivers/uavcan/libuavcan' (https://github.com/PX4/libuavcan.git) registered for path 'src/drivers/uavcan/libuavcan'
49.21 Submodule 'src/drivers/uavcan_v1/legacy_data_types' (https://github.com/PX4/public_regulated_data_types.git) registered for path 'src/drivers/uavcan_v1/legacy_data_types'
49.21 Submodule 'src/drivers/uavcan_v1/libcanard' (https://github.com/UAVCAN/libcanard.git) registered for path 'src/drivers/uavcan_v1/libcanard'
49.22 Submodule 'src/drivers/uavcan_v1/public_regulated_data_types' (https://github.com/UAVCAN/public_regulated_data_types.git) registered for path 'src/drivers/uavcan_v1/public_regulated_data_types'
49.22 Submodule 'src/drivers/uavcannode_gps_demo/libcanard' (https://github.com/UAVCAN/libcanard.git) registered for path 'src/drivers/uavcannode_gps_demo/libcanard'
49.22 Submodule 'src/drivers/uavcannode_gps_demo/public_regulated_data_types' (https://github.com/UAVCAN/public_regulated_data_types.git) registered for path 'src/drivers/uavcannode_gps_demo/public_regulated_data_types'
49.22 Submodule 'src/lib/crypto/monocypher' (https://github.com/PX4/Monocypher.git) registered for path 'src/lib/crypto/monocypher'
49.22 Submodule 'src/lib/ecl' (https://github.com/PX4/PX4-ECL.git) registered for path 'src/lib/ecl'
49.22 Submodule 'src/lib/matrix' (https://github.com/PX4/PX4-Matrix.git) registered for path 'src/lib/matrix'
49.22 Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/Micro-CDR.git) registered for path 'src/modules/micrortps_bridge/micro-CDR'

------ I have removed a large portion of this terminal output, as it did not contain any errors/warning messages ------

338.6 [ 69%] Building CXX object src/modules/vmount/CMakeFiles/drivers__vmount.dir/input_test.cpp.o
338.9 [ 69%] Building CXX object src/modules/vmount/CMakeFiles/drivers__vmount.dir/output.cpp.o
[... cut ...]
ERROR: failed to solve: process "/bin/sh -c git clone https://github.com/PX4/PX4-Autopilot.git &&    cd PX4-Autopilot &&     git checkout v1.12.3 &&     git submodule update --init --recursive &&     make -j8 px4_sitl_default gazebo &&     make px4_sitl_default install &&     cd &&     rm -rf PX4-Autopilot" did not complete successfully: exit code: 2
```

I want to test this out in the simulation first, and then hopefully implement this on a Qualcomm RB5 drone. Any help will be appreciated.

Thanks!

MatWac56, 2024-07-24:
Hello !

I also had this problem and solve it by adding _HEADLESS=1_ before _make -j8 px4_sitl_default gazebo_ :

RUN git clone https://github.com/PX4/PX4-Autopilot.git &&\
    cd PX4-Autopilot && \
    git checkout v1.12.3 && \
    git submodule update --init --recursive && \
    HEADLESS=1 make -j8 px4_sitl_default gazebo && \
    make px4_sitl_default install && \
    cd && \
    rm -rf PX4-Autopilot

guptaPrabhav, 2024-07-24:
Hello @MatWac56 

This solution worked! Thanks a lot for your quick response!

closed 2024-07-24 by guptaPrabhav