matlabbe/rtabmap_drone_example #22: Real drone

Murali25111993, 2024-11-29:
Hello,

I have tested the example in real drone, My drone is changing to offboard mode, rtabmap is generated. I can see the /mavros/vision_pose/pose topic when i run offboard example. But , upon taking off it is moving above the pre-defined limit **1.5 m** and keep on increasing. I dont know why!. 

one thing I did on  the drone is removed GPS, since it is Indoor testing.

Hardware setup:

Nvidia Orin Nano
ubuntu 20.04
px4 1.12.3

I will share any logs if required. 

Any inputs , on how to resolve this. 

**Note:**
I have changed the offboard code such that it publishes only /vision_pose/pose and commented the arming takeoff commands. And tested manually lifting the drone  in x y z axis to check whether /pose value is changing appropriately or not. It is perfectly fine. 

Thanks

Murali

matlabbe, 2024-12-02:
Maybe you could compare the SetPoint and the current position estimated by px4 through mavros topics and see if it is following `/vision_pose/pose` topic.

I don't remember exactly, but it is also possible to visualize current px4 data through QGroundControl to debug more why the drone is not stopping on the set point.

Murali25111993, 2024-12-02:
Yes,

As i can see the local Position z graph, Z set point is in -0.5, the Z estimated is starting from 0. 

[image]

Also you mean I need to monitor the below two topics.? 

[image]

matlabbe, 2024-12-03:
Actually, just remembered it, you can post your px4 log here https://review.px4.io/ and post the link so we can see the full data.

I think if you are based on the offboard example, mavros should republish the vision pose (which is ENU) in the right coordinate frame used by px4 (NED). So a setpoint of -0.5 is indeed 50 cm over the ground. Not sure why the drone is going down (+z) in your plot. The full px4 log may help to understand why.

Murali25111993, 2024-12-03:
I have attached the log file in zip file below.

Also I am attaching the [log analysis link](https://logs.px4.io/plot_app?log=d7967d8c-e4f3-45bb-b788-c92425c28d9d)

[log_83_2024.zip](https://github.com/user-attachments/files/17987182/log_83_2024.zip)

matlabbe, 2024-12-03:
This is the log from the simulated demo of this repo, just taking off then land a couple of seconds after:
https://review.px4.io/plot_app?log=0eb60080-deca-450c-b937-284e66f829c5

Ignoring the manual part (which I think is when the simulator is starting up), there is the ideal offboard curve of the take off at 1.5 m then landing shortly afterwards.
[image]

In your flight log, there is nothing about Visual Odometry like this:
[image]

From the log we can also see that the EKF filter started to integrate vision pose when I started offboard node:
[image]

If you don't see same logs, maybe something is not configured correctly on the px4 controller. Make sure the EKF settings are correctly set in the configs with QGroundControl.

Murali25111993, 2024-12-03:
Ok I understood, I will dig more into this.

I am attaching the existing params. 
[vision.zip](https://github.com/user-attachments/files/17987611/vision.zip)

 What are the config I have to change, any link?

matlabbe, 2024-12-03:
I would have said parameters like:

https://github.com/matlabbe/rtabmap_drone_example/blob/3055aea3a69da6c8acc0177c4c9c386578df38d5/config/1099_iris_rtabmap#L11-L12

but it seems you are setting them already. Not sure why we don't see any logs about the EKF though.

Murali25111993, 2024-12-03:
I will explore more into this . I will keep you posted about development, meanwhile, if you have any more suggestion please update me. 

Thanks

Murali25111993, 2024-12-04:
Atlast it worked, made it to hover at 0.5 m. 

I am attching the [log](https://logs.px4.io/plot_app?log=f8020f09-cd7d-4028-becd-5281edd3ea7c)

attaching the params 

[working_vision.zip](https://github.com/user-attachments/files/18006186/working_vision.zip)

Now, I will move on to mapping then navigation . 

Thanks, this issue can be closed now.

matlabbe, 2024-12-07:
Great that you've found a solution and shared it!

closed 2024-12-07 by matlabbe